Spacecraft-NX/firmware #3: .Failed to write! 6 and Failed to read! 6

froid-san, 2020-12-25:
Switch SX-core Erista patched flashing using windows 10 64bit

keeps getting an error .Failed to write! 6 and Failed to read! 6 when clicking flash.bat

Gateway DFU found!
Flashing bootloader updater.................................................................................................................................................................................
Waiting for the updater to boot...
Failed to open! 2
Backing up the current bootloader.
.Failed to write! 6
Failed to read! 6
.Failed to write! 6
Failed to read! 6
.Failed to write! 6

...
Failed to write! 6
Failed to read! 6
Flashing the new bootloader.
...
.Failed to write! 6
Failed to read! 6
.Failed to write! 6
Failed to read! 6

Failed to write! 6
Spacecraft-NX DFU not found!
!!!DONE!!!
Press any key to continue . . .

Then the modchip won't work and I'll just boot to ofw  sysnand (I got emunand).

flashed the backup bootrom and the SXOS v1.3 firmware using the sxupdater_win32 and everything is back to normal. 

Tried 2 different USB cable, different USB port (USB 2.0/3.0) and a newly installed OS on a newly build mini PC with the modchip is the only port detected.

greckerdli, 2020-12-28:
What is there, I have the same problem only that the boot did not backup well and now I cannot recover the chip, I don't know if you could pass me the bootloader file of your backup to see if that solves my problem, thank you in advance.

balika011, 2020-12-28:
If the backup wasn't made then the bootloader wasn't flashed over.

froid-san, 2020-12-28:
> What is there, I have the same problem only that the boot did not backup well and now I cannot recover the chip, I don't know if you could pass me the bootloader file of your backup to see if that solves my problem, thank you in advance.

as balika011 said maybe it isn't flashed yet, try flashing the sxcore v1.3 firmware 1st using the sxupdater_win32 and see if it recovers your firmware.

I don't have any fancy antivirus software and just using the plain old windows defender build-in on windows. 

btw I already tried it running on administrator mode via cmd. and still got the same error.

greckerdli, 2020-12-28:
sxupdater does not recognize it, it tells me that sxcore / lite not detected, I will try to modify the bat to see if it only installs the spacecraft bootloader

topfire666, 2020-12-28:
I too fail 6.

froid-san, 2020-12-30:
> sxupdater does not recognize it, it tells me that sxcore / lite not detected, I will try to modify the bat to see if it only installs the spacecraft bootloader

I can share my bootloader.bin but I need balika011 confirmation if it would be safe to do so for your modchip, as it may do kore damage.

topfire666, 2020-12-30:
Could you force flash without backup?  Since our bootloader seems to be corrupt.

topfire666, 2021-01-01:
hi, i managed to flash the chip, changing computers.

dariodimarcoberardinoRetroHax, 2021-01-08:
> > sxupdater does not recognize it, it tells me that sxcore / lite not detected, I will try to modify the bat to see if it only installs the spacecraft bootloader
> 
> I can share my bootloader.bin backup but I need balika011 confirmation if it would be safe to do so for your modchip, as it may do kore damage.

you mind sharing it with me too since mine completely fucked and well gives me the error failed 6 too

topfire666, 2021-01-08:
I repeat, I tried flashing with another pc, I also got error 6 and I tried it on a laptop with windows 10 i7 and I flashed it the first time.  Nothing has been flashed, it is just locked.  And when you flash it, you will have your correct backup.

balika011, 2021-01-08:
This happens when a program opens the device before the updater does and keeps it open. Only one software can handle COM devices at once on Windows.
TODO: Add better message and possible print what software keeps it open, so it can be closed or uninstalled.