Spacecraft-NX/firmware #8: Add an option to disable glitching (à la Genuine Boot from TX)

lordfriky, 2021-01-06:
I don't want to depend on Atmosphère every time I want to boot into OFW, I just want to have an option to disable the glitching just like the *Genuine Boot* option from SX OS (I know it wasn't a clean boot since TX shit was still written on the NAND, but at least I didn't have to update the CFW every time a firmware update broke support for it).

balika011, 2021-01-06:
"Genuine Boot" is not possible with this chip. They lied to you and boot modified firmware when they say is "Genuine".

closed 2021-01-06 by balika011

balika011, 2021-01-06:
Also, what you want to achieve is a bootloader issue.

Hekate has ifdef'd support for booting real-stock on mariko, you could compile hekate with this define on and it should do what you want: https://github.com/CTCaer/hekate/blob/4a152504cb3aeeee32f500ea6f36d7038501f880/bootloader/hos/hos.h#L46