Spacecraft-NX/firmware #9: Can't reflash SX firmware.

ludufre, 2021-01-06:
I installed Spacecraft-NX to dump my partialseas.key on my Mariko with SX Core.

Now, I want to go back to SX firmware and the installer says:

[image]

It's not serial issue:
[image]

balika011, 2021-01-06:
Ask them

closed 2021-01-06 by balika011

ludufre, 2021-01-06:
No sir! Your firmware changed something. I've previously updated from 1.0 to 1.3 with same software. The problem started after flased your firmware. 

If you don't even want to bother to investigate. It would be fair to include this alert in README.md, because I am already in the second Mariko who does not accept to return to SX after installing your bootloader / firmware.

ludufre, 2021-01-06:
@balika011 I solved. I run .\BootloaderUpdater.exe .\backup_2021_01_06_06_13_13.bin and now the Sx Updter recognizes.

This is worth adding to README.md. Can I provide a PR?

balika011, 2021-01-06:
No