Spacecraft-NX/firmware #10: Flashed...but not working & can,t rollback

MonsieurKimYong, 2021-01-07:
When flashing the chip, It was well done with backup files.

But  when I turn on the switch, boot on genuin with white LED  light on SXchip.

So I tried to restore SX firmware with Spacecraft & Sxupdater...

It say "no device detected" 

when I connect with my 64bit Windows10, It's LED is green light.

What's the cause? How can I fix it?   

1 SXcore & 1 SXlite same situation. PLZ  help me.

topfire666, 2021-01-07:
If you go back to trying to flash, it doesn't give you error 6 or block?  The light is blue and when you connect it to Windows it turns green.  I solved it simply using another pc.
Open backup and i Will see corrupt file.

samielis, 2021-01-08:
im having the same issue. any ideas?

balika011, 2021-01-08:
Turn off the console by holding the power for 12 seconds and check your usb connection.

edkozuto, 2021-01-08:
I have encountered the same issue.
Now the switch just boots straight to an untouched sysnand.

MonsieurKimYong, 2021-01-09:
Try with another computer. I solved the problem ^^

*KEB하나은행 476 91017 821307 김경용 *

*받으신 게임이 설치 및 실행에 이상이 있는 경우, 파일의 문제가 아니라 펌웨어 버전의 문제가 대부분입니다.  증상과 함께 기기의
펌웨어버전을 알려주세요. 스위치의 '설정>본체>본체업데이트' 아래 작은 글씨로 써 있습니다.*

*추가공지 - 정펌, 커펌 둘 다 사용가능하게 에뮤낸드(멀티부팅) 작업하신분도 이제는 파일과 메뉴얼 받아서 직접 업데이트가 가능합니다.
단 멀티부팅 작업 및 신형 코어 작업하신 기기시라면 그 내용을 말씀주셔야 추가적으로 하셔야 할일을 안내드리니 작업전 안내 받으시고
업데이트 진행하세요~~~*

*다운로드 용량 초과 오류시 해결방법 : https://extrememanual.net/33752
<https://extrememanual.net/33752>*

*꼭 성함, 전번, 펌웨어버전 꼭 넣어서 메일로 신청 부탁드리고, 자료요청 관련하여 전화, 톡, 문자 사절입니다. *

*메인 블로그를 티스토리로 이전했습니다.  kky111111.tistory.com
<http://kky111111.tistory.com> 방문하셔서 구독 및 댓글 하나씩 남겨주시면 큰 힘이 되겠습니다. 게임관련 언급은
절대 마시구요 ^^"*

*다운로드 완료되시면 입금 부탁드리고 바로, 꼭 "입금 및 다운완료" 답메일 남겨주세요. *
*미리 말씀없이, 공유드린 시점에서 24시간 이내에 다운완료 확 메일이 없을 경우 임의로 자료 내립니다. 또한 반복적으로 그러시는** 분들은
저도 요청에 답 안드릴테니 서운해 마세요. ^^"*

samielis, 2021-01-09:
Switching to my laptop fixed the issue. Still, not really a satisfying solution.
Is any body experiencing the issue using Kaspersky Security Software?

edkozuto, 2021-01-09:
> 
> 
> I have encountered the same issue.
> Now the switch just boots straight to an untouched sysnand.

Fixed on another PC, too

topfire666, 2021-01-10:
I have another chip to flaw, I have tried to install windows 10 clean home and pro and it keeps giving the same error.

sthetix, 2021-01-26:
use zadig to find the device, click options, list all devices, and if it is detected by ur windows, the modchip will be seen as DFU. make sure to upgrade or change its driver to usb serial cdc.

cryptblood1986, 2021-05-01:
> use zadig to find the device, click options, list all devices, and if it is detected by ur windows, the modchip will be seen as DFU. make sure to upgrade or change its driver to usb serial cdc.

Did that. Still no luck. 
Gets stuck at creating backup bootloader.
I've tried 8 usb cables.
One computer and a laptop.

sthetix, 2021-05-01:
Try using other usb port. I had that problem too, but I've fixed it by
plugging the chip to other USB port.

cryptblood1986, 2021-05-01:
> Try using other usb port. I had that problem too, but I've fixed it by plugging the chip to other USB port.
> […](#)

Thanks that actually worked. Log says successfully flashed but when I turn on the unit I'm getting a blinking green light with no screen. Is this normal?