Spacecraft-NX/firmware #12: Stuck on Solid Green LED

Metron7757, 2021-01-10:
LED pulses a pinkish color then will turn green. One of 2 things happens when it turns green. 1: Light turns Green and stays green “Black Screen” or 2: Light turns green then Immediately shuts off and boots Hakate. Does this without a SD card also. 6 out of 10 boot ups get stuck on this green light.

balika011, 2021-01-10:
Is this an Erista? There is a known issue where sometimes the glitch corrupts the memory in the bootrom. It loads the payload, the modchip detects it booted, but the bootrom crashes and the payload doesn't start. It happen to me, but seemed too rare to implement a workaround. This never happen on Mariko.

Metron7757, 2021-01-10:
Yes I have a Erista Unit, thanks for the feedback. Will keep using as it's not damaging anything. Keep up the awesome work! So glad to be free from SX OS.
Thank you 👏🏻

ktobyznal, 2021-01-24:
Have the same issue with my Erista. Never patched.
The Light stays green and “Black Screen” every time.

MEMFlasher98, 2021-01-24:
My patched Erista is like that and now its rare to my switch to turn on PLEASE HELP!

sthetix, 2021-02-04:
It happened to me for the lite (the one with weird sleep function) , two times black screen, and solid green led.

MEMFlasher98, 2021-02-04:
> It happened to me for the lite (the one with weird sleep function) , two times black screen, and solid green led.

so what do i do?

sthetix, 2021-02-05:
> > It happened to me for the lite (the one with weird sleep function) , two times black screen, and solid green led.
> 
> so what do i do?

I dont remember how i fixed it but I tried to remove the battery and replug it.

NarakuYume, 2021-02-12:
Got the same problem on my erista and it often shows me a black screen. Is there any solution?

sthetix, 2021-02-13:
Usually you need to try to boot it up again. it will go through

men9xuan, 2021-02-15:
Try boot into emunand and reflash a system update at your current system version. I only tried on emunand and I think it won't hurt to relash a sysnand upgrade as well. TLDR  I experienced black screen before trying to boot into emmunand. My guess is SX Core modified the sysnand's BOOT1 and BOOT0 and the emunand created from it also has modified BOOTs, so it wont work properly with hekate or fussee primary. After  reflash the system, I assume it was reverted back to stock and thus works well with spacecraft/hekate/fussess.

Katzenwahn, 2021-02-16:
I wanted to test out Spacecraft-NX on my Erista so I flashed it on my modchip and after the training session, I only got a black screen instead of the "NO SD" logo. I was a little bit afraid about this and I thought, I did something wrong but luckily I did a research and then I found this here. As far as I understood, this will only occur on Erista consoles? So I turned it off and pressed the power button again and then the logo appeared and I was relieved. After that I tested it out a few times to see how often this black screen occurs. In my case, it was like a 50 percent chance to boot successful into the payload or to get the black screen. That was a little bit annoying for me and so I decided to flash back the original firmware of the modchip. If this issue will be fixed with an update, then I will give Spacecraft-NX another try. :)

men9xuan, 2021-02-16:
> I wanted to test out Spacecraft-NX on my Erista so I flashed it on my modchip and after the training session, I only got a black screen instead of the "NO SD" logo. I was a little bit afraid about this and I thought, I did something wrong but luckily I did a research and then I found this here. As far as I understood, this will only occur on Erista consoles? So I turned it off and pressed the power button again and then the logo appeared and I was relieved. After that I tested it out a few times to see how often this black screen occurs. In my case, it was like a 50 percent chance to boot successful into the payload or to get the black screen. That was a little bit annoying for me and so I decided to flash back the original firmware of the modchip. If this issue will be fixed with an update or with a confirmed another solution, then I will give Spacecraft-NX another try. :)

Oh I didn't notice black screen during "NO SD" logo. It seldomly failed to boot with led blinking for a long time. I would hold the power button for 10s to kill the power sending to the chip and power on again. This usually does the trick. Good luck on your journey.

Katzenwahn, 2021-02-16:
> Oh I didn't notice black screen during "NO SD" logo. It seldomly failed to boot with led blinking for a long time. I would hold the power button for 10s to kill the power sending to the chip and power on again. This usually does the trick. Good luck on your journey.

Thank you for your comment, I guess this issue was similar like the user "Metron7757" described it here because he said, "6 out of 10 boot ups get stuck on this green light" and I felt the same.

NarakuYume, 2021-02-23:
> Try boot into emunand using fusee secondary and reflash a system update at your current system version. I only tried on emunand and I think it won't hurt to relash a sysnand upgrade as well. TLDR I experienced black screen with erista console trying to boot into emmunand. My guess is SX Core modified the sysnand's BOOT1 and BOOT0 and the emunand created from it also has modified BOOTs, so it wont work properly with hekate or fusee primary. After reflash the system, I assume it was reverted back to stock and thus works well with spacecraft/hekate/fusee.

I tried that out and flashed my emunand and sysnand and still get a black screen sometimes when I boot up the console. I noticed another thing in NXNandMangager that my BOOT0 has enabled AutoRCM and I don't know how it got activated.  Should I disable it in NXNandManager? I thought AutoRCM is only for unpatched units?

men9xuan, 2021-02-25:
> > Try boot into emunand using fusee secondary and reflash a system update at your current system version. I only tried on emunand and I think it won't hurt to relash a sysnand upgrade as well. TLDR I experienced black screen with erista console trying to boot into emmunand. My guess is SX Core modified the sysnand's BOOT1 and BOOT0 and the emunand created from it also has modified BOOTs, so it wont work properly with hekate or fusee primary. After reflash the system, I assume it was reverted back to stock and thus works well with spacecraft/hekate/fusee.
> 
> I tried that out and flashed my emunand and sysnand and still get a black screen sometimes when I boot up. I noticed another thing in NXNandMangager that my BOOT0 has enabled AutoRCM and I don't know how it got activated. Should I disable it? I thought AutoRCM is only for unpatched units?

Can you post your hekate config file? I don't think AutoRCM is useful for erista. Never encontered this issue.

Tur0l, 2021-02-26:
omg i'm so excited right now.
My console is an erista patched with firmware 11.0.1 and when I flashed Spacecraft-NX I was stuck with the green led.

Not being able to go back to SX OS (it didn't boot), I tried to flash Spacecraft-NX again with the same result.
What was my surprise that by holding "Vol +" for 3 seconds and then the power button, I boot hetake without any problem.

I have repeated it several times with success.

hope this helps the rest of erista patched with SX core.

Metron7757, 2021-02-26:
Interesting can you do a test for me try booting the Console￼ and out of the 10 times￼ how many successful boots is it able to do?￼￼

men9xuan, 2021-02-26:
> Interesting can you do a test for me try booting the Console￼ and out of the 10 times￼ how many successful boots is it able to do?￼￼
I can boot 10/10.

Metron7757, 2021-02-26:
And can anyone else verify that this works with there Erista unit? Don't currently have access to my Switch so I can't test this claim￼

Tur0l, 2021-02-26:
I do not know what is happening. It may have been luck but after doing numerous tests I have obtained:

pressing power 15 of 20

pressing vol + + power 18 of 20

pressing vol - + power 1 of 10

NarakuYume, 2021-02-26:
I tested it with holding Vol + and it doesn't work for me, I still get a black screen. I guess it's completely random to boot successfully into the payload. The only solution is not to turn off the console anymore and have it in standby mode.

gudenau, 2021-04-18:
This also occurs on my "unpatched" Switch, just rarely.

kennykdash, 2021-06-12:
I need help. I had chip flash with spacecraft nx, initially everything was fine until the atmosphere error logo shows up during gameplay. Now the chip will blink purplish, then stuck on green with black screen. Trying to replace the chip, it did enter to sx logo no SD. however connecting the chip via micro usb to my pc will only shows no device detected. Tried zadig, it doesnt show any device detected. so i have two issues basically. Summary
1. initally spacecraft was flashed sucessfully, then error during gameplay, now blinking purplish then solid green with black screen
2. now i have second chip. my pc cannot detect and i cannot flash sx to spacecraft nx. 

appreciate if anyone could help me figure out what is happening or what went wrong

men9xuan, 2021-06-12:
> I need help. I had chip flash with spacecraft nx, initially everything was fine until the atmosphere error logo shows up during gameplay. Now the chip will blink purplish, then stuck on green with black screen. Trying to replace the chip, it did enter to sx logo no SD. however connecting the chip via micro usb to my pc will only shows no device detected. Tried zadig, it doesnt show any device detected. so i have two issues basically. Summary
> 
> 1. initally spacecraft was flashed sucessfully, then error during gameplay, now blinking purplish then solid green with black screen
> 2. now i have second chip. my pc cannot detect and i cannot flash sx to spacecraft nx.
> 
> appreciate if anyone could help me figure out what is happening or what went wrong. my switch is mariko v2

I highly suspect that the sd card connection is not installed properly. You need to check the flex cable connector is fully pressed down.

kennykdash, 2021-06-13:
Great i managed to reflash back the previous chip.  Solution
1. Reflash back the chip using v1.3 sx firmware. Then flash back to spacecraftnx. The error causing is unknown, im not sure if it will reoccur again. At least my existing chip is still usable.
2. Pc cannot detect DFU mode. Have to restart my pc few times. Probably driver issues

kdgbduyu28, 2021-09-16:
Does anyone have the file sxcore_fw_v1.3.bin? Xcecuter site is now inaccessible. Thank you all!

sthetix, 2021-09-16:
https://u.pcloud.link/publink/show?code=XZwhrYXZLixF0Dk6KeJDWzcLhP2Y7B8pdc7X

try that