Spacecraft-NX/firmware #15: Always blinking magenta

HandsomeGleb, 2021-01-21:
I have a problem with the sx lite.
I install spacecraft-nx on my sxlite, and when I turned on switch it started blinking green/yellow and blinking goes soo long, i’ve waited about 2-3 hours then switch turned off. When i turn it on again it starts blinking magenta, i waited all night and on the morning it still blinking. Then I disconnect sx lite and try to restore my backup with sx, when i plug the chip back and turned on my switch it start blinking blue forever. 
Then, i installed spacecraft on the chip again, and the problem of magenta blinking come with it. Wtf? Does i bricked my chip, or what?

tuxuser, 2021-01-24:
It could be simply related to the soldering not being 100% proper. So the glitching process never succeeds.
Did you test the glitching process for success before you installed Spacecraft?

HandsomeGleb, 2021-01-24:
> It could be simply related to the soldering not being 100% proper. So the glitching process never succeeds.
> 
> Did you test the glitching process for success before you installed Spacecraft?

Yea, it works great with sx os before

fille1976, 2021-04-30:
This just bricked my sxlite,what a piece of shit software is this,failed to wright,failed to read,i let the program do till its says done,and now it doesn't recognised anymore in windows,and flash first white then green,thx,i followed how it must be done.

sthetix, 2022-03-07:
> having the same problem, it is stuck blinking magenta. didnt have any
> problems before with the original sx firmware

you must unbrick it using the gd link programer or the st link. i have a
guide to do that on my yt channel https://youtu.be/RK5FQZe1MAs

T-Boa98, 2022-03-07:
> you must unbrick it using the gd link programer or the st link. i have a guide to do that on my yt channel https://youtu.be/RK5FQZe1MAs
> […](#)

I will try that. Since I removed the chip from my switch. Do I first have to resolder the sx core back into my switch or does the unbrick work only with the chip itself connected to a pc?

sthetix, 2022-03-07:
Unbrick without the need to install the chip to the console.

T-Boa98, 2022-03-07:
> 

okay, thank you. I´ve also noticed that you are using a different Modchip. I have the original SX Core Modchip for a mariko Switch. Where do i have to install the jumper cables there?

sthetix, 2022-03-07:
[image]
Please use this guide.

sthetix, 2022-03-07:
[image]
[image]

T-Boa98, 2022-03-07:
> [image] [image]

Thank you very much for the fast help!
One last question when i rewrite the bootloader to the chip and i want to use Spacecraft-NX as the firmware. What do i have to write to the chip? The bootloader.bin or the firmware.bin?

[image]

sthetix, 2022-03-07:
> <https://user-images.githubusercontent.com/22261135/156998551-3ed6e76b-e7d0-4fe6-a857-6b6aeb86e9e5.jpg> [image:
> sxcore-pinout]

bootloader at offset 0x08000000 and firmware at offset 0x08003000

T-Boa98, 2022-03-07:
so basically i have to write both .bin onto the chip? do i have to do that in two seperate programmming steps or can i chose the bootloader at its offset and the firmware at its offset in one programming step?

sthetix, 2022-03-07:
same programming but flash it on different offset

T-Boa98, 2022-03-07:
okay, i´ll try to follow everything you said. But first i have to buy the GDLink OB and the cables. Thank you very much for your advices!

T-Boa98, 2022-03-07:
> same programming but flash it on different offset
> […](#)

can i also just follow your tutorial

https://www.sthetix.info/unbricking-the-tx-lite-and-tx-core-modchips-updated-version-easier-to-do/

this seems to be the correct one for me? But in this one you flash the bootloader via the gd-link ob and the nx firmware via the cmd command. si is it better to make it like u did in the video or like you said both via the gd-link ob with the different offsets?

sthetix, 2022-03-07:
it is easier to flash the bootloader and firmware on separate offsets using
the same program. fyi my video is old but still works..