Spacecraft-NX/firmware #18: SD card us no longer detected

Romain12362, 2021-02-03:
Hello, I have a Mariko unite, I had no problem with my SD card before starting the changes.  After flashing Spacecraft-NX, my Switch did not want to detect my sd card (I stay on a "no sd" screen).  I changed SD card, same problem.  I reflash the old bootloader and the old firmware, the console boot up and asks me for the "boot.dat" file but the SD card is still no longer recognized.  I changed the SD card reader but the problem is still there, what could have gone wrong?

sthetix, 2021-02-03:
i had that problem too, check the sdcard socket. test all its pins.

Romain12362, 2021-02-03:
> i had that problem too, check the sdcard socket. test all its pins.
> […](#)

 Hi, thanks for your answer.
My multimeter is broken so I bought a new one and wait to receive it.
When I got it and test it, I will tell u how it done.

Romain12362, 2021-02-05:
Hello,
My multimeter connectors are too fat.
So I'm not sure to have something on all the pins of sd card reader.
Is it normal? 
Is it the chipset flash who did it?
Tx for your answer

closed 2021-03-07 by balika011