Spacecraft-NX/firmware #20: White LED flashing directly with ST Link v2

CabrouX, 2021-02-17:
Hi,
I'm having problems booting the switch having flashed the bootloader and firmware through ST Link V2 and (bootloader in 0x08000000 and firmware in 0x08003000).

When powering on, the SX Core lights up an static white led and boots directly into OFW. 
Do you know why this is happening? Any idea how to make it work?
I don't have the USB port available anymore in the SX Core so  flashing through USB is not a possible option for me.

Thank you!

sthetix, 2021-02-17:
I have a video to unbrick the chip and you cannot use the STLink v2 to do
that. You need GDLink to reflash the chip
you can watch it on my site

sthetix, 2021-02-18:
> Hi,
> I'm having problems booting the switch having flashed the bootloader and firmware through ST Link V2 and (bootloader in 0x08000000 and firmware in 0x08003000).
>
> When powering on, the SX Core [...]

and you can make the usb cable connection yourself. look at my tweeter post to find it.

https://twitter.com/sthetixofficial/status/1359113294452117508?s=19

Thisisntmeasfarasiknow, 2021-03-05:
@sthetix After much messing around I was able to succsessfully write to flash using a J-Link programmer, unfortunately i was not able to use J-Flash program and instead had to do it via Keil uVision (with the programmer selected), after which i was able to successfully read back what i wrote and can confirm the data is indeed what i wrote.... Using a J-Link is much more practical as they're more common and unlike the device specific GD32 programmer it is actually useful more than once for other devices.

Unfortunately following this i was unable to update the firmware/bootloader using the Spacecraft CMDs over USB comms as detailed in your video.  

When you were using Keil, was there anything specific you had to do in order for this to succeed as compared to using the GD software? does it handle the .hex/.bin differently for example etc?

Thanks.

sthetix, 2021-03-05:
you can watch my previous video. i used keil for the first video.

Thisisntmeasfarasiknow, 2021-03-05:
> 
> 
> you can watch my previous video. i used keil for the first video.

Hi, thanks!. Can you provide a link to that video? i tried to find it but kept just getting your updated video

sthetix, 2021-03-05:
Sure, here it is.
https://odysee.com/@sthetixofficial:9/unbrick-sxcore-lite:8?r=2VGbSyEYwC3Z5VkhMrookuMgudDR2qFS

I've reuploaded the video.

Thisisntmeasfarasiknow, 2021-03-05:
Thanks @sthetix I really appreciate that.

While my initial steps were very similar to yours they weren't identical. I'll try again mirroring your steps 1:1 and see if there is a change :)

Thanks again.

balika011, 2021-03-07:
ST Link doesn't support GD32

closed 2021-03-07 by balika011

Thisisntmeasfarasiknow, 2021-03-07:
@balika011 

Just to confirm incase USB Comms don't work on my system for some reason after manually flashing the bootloader with J-Link.

Is CabrouX correct, the firmware offset is 0x08003000 ? and I'll try manually writing it.

Thanks