Spacecraft-NX/firmware #26: Slow SD card is not detected

blawar, 2021-11-29:
micro sd card: PNY 512 Pro Elite

Spacecraft always errors out with "NO SD" with this card.  This same SD card seems slow to initially respond in windows, however it did work on the previous modchip firmware, and it works with AMS on my unpatched erista console.  Perhaps the timeout values are too tight?

Southsystemic, 2021-12-09:
This has happened multiple times with 256gb ONN cards also, only some of them give problems. You can try to turn on the switch without the card and then immediately insert the sd. Sometimes it works.

balika011, 2022-01-12:
Known fusee issue. Can't reproduce with my cards.
If somebody sends me a card that has that issue i'll fix it.

closed 2022-01-12 by balika011