Spacecraft-NX/firmware #28: [Question] Sleep mode issues

DarkenLX, 2022-01-04:
Hi I have a patched Erista had OFW 8.0.1 Originally has 13.2.0 OFW/CFW Atmosphere Deepsea latest/current version as of 01-03-2022 I have spaceCraftNX install latest as of 01-03-2022 but I have been trying to track this issue down if it's hardware or software I am not sure if this might be due to spaceCraftNX or the mod chip itself.. is the weird sleep issue with the "lite" switch known to happen with Erista board too? If so Where can I find info on the cause if known? Thanks.

balika011, 2022-01-12:
What is the weird sleep issue?

DarkenLX, 2022-01-12:
Someone mentioned a weird sleep issue here https://github.com/Spacecraft-NX/firmware/issues/12#issuecomment-773012187 
And something in atmosphere GitHub: https://github.com/Atmosphere-NX/Atmosphere/issues/1357
Deepsea is just a customized packed vanilla atmosphere basically it has added homebrew that comes with it nothing really different.