Spacecraft-NX/firmware #32: BLACK SCREEN.. stuck on green LED

virienko, 2022-01-17:
After a successful update to v3, it hangs on the green LED. Doesn't work with or without SD card. Only black screen. If you forcibly restart the console and immediately turn it on, OFW starts.

Tried version 1 from the repository. The result hasn't changed.

Can you help somehow? Perhaps another update? Thank you

balika011, 2022-01-25:
What is v3?!

MagnusHydra, 2022-02-21:
> What is v3?!
V3 is the cpu ribbon people don't understand that its not the name of the chip nor identify which chip they are using. Sounds like they flash their chip with the wrong firmware type for their chip.

johnkelvinliu, 2025-09-13:
hwfly mod chip, Switch OLED, firmware v20.4.0.
Tried up to 20 times and flashed the v0.2.0 twice.
No matter microSD is inserted or not, I still have black screen and and mod chip green led keep on lights up.
It is frozen afterwards.
When using hwfly-nx v0.7.2, it works normally.