Spacecraft-NX/firmware #34: Sx Core micro-USB loose contact

T-Boa98, 2022-02-19:
Hello, I got the problem that it seems that the SX Core´s Micro-USB Port is quite loose. So I´m unable to flash Spacecraft-NX, since I´m afraid to brick the Sx Core when it loses Connection while flashing. Is there any other way to flash Spacecraft-nx onto the SX Core? or is there a way to get another fittinh Micro-USB port?

Thanks in advance

ryan1109, 2022-02-20:
reinsertion. He'll be fine.

T-Boa98, 2022-02-20:
> reinsertion. He'll be fine.

Reinserted it for hours actually, always with the same result

MagnusHydra, 2022-02-21:
That has nothing to do with the firmware...

carcaschoi, 2022-02-21:
replace with hwfly usb port

T-Boa98, 2022-02-21:
> replace with hwfly usb port

Any idea where to get one without buying the whole chip?

T-Boa98, 2022-02-21:
> That has nothing to do with the firmware...

That's why I'm asking if there is a way to flash it without using the sx cores USB port.

rusagent, 2022-03-07:
I have the same problem and a solution would be super awesome <3

sthetix, 2022-03-07:
please take a look at my post on Twitter

https://twitter.com/sthetixofficial/status/1359113294452117508?t=ReCJ5lDXBlwW8Nb-2eAX4Q&s=19