Spacecraft-NX/firmware #37: 0.6.0 HWfly OLED glitch fails when updating OFW to 14.0

dryja123, 2022-04-04:
Just ran into an issue where the HWfly OLED does not want to glitch directly after updating OFW to 14.0.  I updated all CFW and updated the firmware on my emunand to 14.0 using Daybreak just fine.  Tested multiple reboots and was able to launch Hekate and Atmosphere into my emunand.  

I booted into OFW and updated the firmware to 14.0.  When the Switch went to do the reboot the mod chip would not glitch again.  I would get a purple pulsing light for approximately 2 minutes and then a solid amber / yellow light.  I rebooted again with the same results.  The third time I rebooted I got a purple pulsing light for approx 2 minutes and then a solid red light.  

I opened the console and checked all of my solder connections in diode mode and all were testing with the proper resistance.  I went to try to update the mod chip to 0.6.2 and received a message that the bootloader and firmware failed to update.  I factory reset the mod chip and the Switch booted into OFW 14.0 again just fine.  I rebooted again, the mod chip re-trained, and then was booting into Hekate just fine again.  I rebooted several times after and the chip is glitching again almost instantly.

vegedb, 2022-04-04:
>I booted into OFW and updated the firmware to 14.0.

Did you update it through,

1. cfw/daybreak 
2. ofw nint3nd servers through hekate
3. ofw nint3nd servers bypassing hekate

dryja123, 2022-04-04:
> > I booted into OFW and updated the firmware to 14.0.
> 
> Did you update it through,
> 
> 1. cfw/daybreak
> 2. ofw nint3nd servers through hekate
> 3. ofw nint3nd servers bypassing hekate

For the OFW update, I booted into Hekate and selected reboot -> OFW.  I then followed the stock Nintendo firmware update process to update the firmware.  

Keep in mind that the mod chip was glitching fine with the firmware being on 14.0 on my emunand. The glitch broken when the Switch rebooted to do the OFW firmware update.

dryja123, 2022-04-04:
> > stock Nintendo firmware update process to update the firmware.
> 
> This was probably the issue. Using CFW to update to OFW 14.0 using daybreak does not require re-training.

Does using daybreak on your sysnand w/ CFW carry a ban risk?  I used daybreak on my emunand just fine and the issue with glitching happened when I updated my sysnand by regular means.

vegedb, 2022-04-04:
>daybreak on your sysnand w/ CFW carry a ban risk?

Probably.

balika011, 2022-06-11:
hwfly chips aren't supported

closed 2022-06-11 by balika011